commaai/neo #5: board testing: not downloading

xinlutang, 2017-01-03:
We built a NEO circuit board and tested no short in the step one in the page 16 in guide.pdf. I followed the instruction. but in step 4, I connect a 12 V adapter without plugging into a car. I also measured the 3.3V and 5.0 V test pins and all normal. When I hold on SW1, the voltage of BOOT0 change from 0 volt to 2 volts. However, I hold there for a long time, the red LED never flash. On the phone GUI, it say not board found. I did hold SW1 before I apply the power.

On the phone side, I ran it with and without the continue.sh. they is all the same. Could you tell me how long I shall hold on the sw1? I assume couple seconds shall be enough. What is other indicator I can use to check if the board is working?

edsammy, 2017-01-03:
Try testing the board without the 12V supply and just power over USB from a computer 
1. hold down the button
2. with a USB extension cable, plug the board into the computer (while still holding down the button)
3. run 'lsusb' and you should see
`Bus 001 Device 012: ID 0483:df11 STMicroelectronics STM Device in DFU Mode`

This indicates that the microcontroller is live

xinlutang, 2017-01-04:
Red LED flashing now. One of my college changed the manager.py, which caused the problem.

Do I need to enable the start_all env parameter?
Thanks,

geohot, 2017-01-04:
No, the services should start when it detects that you start your car.

tsboost, 2017-01-11:
Can I sell about 1000pcs for eBay for people who are having trouble with neo board production?
If there are no legal problems, we will make neo boards and sell them at a reasonable price.

geohot, 2017-01-11:
You certainly won't have legal problems from comma.ai, as the board is MIT licensed.

gg4u, 2017-01-11:
Wow. Reading about this form Europe. Before starting, are the instructions meant only for  Qualcomm / Snapdragon 820 chips or also for other vendors? E.g. Huawei / Kirin . 
Which are constraints if to adapt? (not an electric engineer here).

@tsboost if it may be a business of yours, you might want to allow boards to communicate with other chipsets.

geohot, 2017-01-11:
Board works with PC and other phones as well. 100% open source.

Adapting off 820 is pretty hard. You can get a OnePlus 3 in Europe.

tsboost, 2017-01-12:
@geohot Thank you for answer. Do you have a Neo Board update or upgrade plan?
@gg4u  Thanks. advice..

closed 2017-01-20 by edsammy