commaai/neo #7: board  found  ,  not downloading

leo333, 2017-01-20:
Try testing the board without the 12V supply and just power over USB from a computer

Perform the following:

hold down the button
with a USB extension cable, plug the board into the computer (while still holding down the button)
run 'lsusb' and you should see
Bus 001 Device 012: ID 0483:df11 STMicroelectronics STM Device in DFU Mode
This indicates that the microcontroller is live

 hold down SW1 On the phone GUI, it say board not found. the red LED never flash. when plug the breakout cable into the vehicle (or 12V adapter), green led flash once

Not hold down SW1 On the phone GUI, it say board found.

edsammy, 2017-01-20:
If it says board found then all is working correctly. Is the red LED still not flashing? If so then check for shorts near the LED

leo333, 2017-01-22:
Red LED flashing now.    On the phone GUI board found.
The services should start when it detects that you start your car. is that so?

edsammy, 2017-01-23:
Correct

edsammy, 2017-02-22:
Anymore issues here?

leo333, 2017-02-22:
TKS  fixed。
I have a question, why not use Solid Cap  16v 330uf，I don't know much about the circuit。

edsammy, 2017-04-03:
If by solid you mean a ceramic capacitor like [this](https://i.stack.imgur.com/qTaTC.jpg) then the answer is because a 330uF ceramic capacitor is very expensive. They usually have much lower capacitances than their electrolytic counterparts (like the one we used).

closed 2017-04-03 by edsammy