commaai/neo #8: how to flash this“neos_v2.zip”

WanXiangGG, 2017-03-12:
my device is OnePlus3  but idont konw how to flash the “neos_v2.zip”to my phone 
 
my recovery is TWRP 3.0.4 
 
i use this recovery to flash this package ，but recovery tell me :"can't find META-INF/com/google/android/update-binary from zip "
 
so please tell me how to dealwith this problem .
 
thanks a lot!

KonradIT, 2017-03-12:
You're supposed to unzip the file in your PC and run unlock.sh or flash.sh file, if you want to do it thru recovery then unzip the file in your phone, flash system.img in TWRP (Flash > Image > Select System > FLASH), Same for boot.img (select boot in TWRP), then wipe cache+dalvik+sd card. (I personally have not tried this).

Or, you could just enable adb in your phone and run flash.sh

WanXiangGG, 2017-03-13:
But now, i find a new question.

i flashed it on my phone , the phone is power on  now.

and screen tell me  i need to "enter the URL of the NEOS program to install"

but i dont know what is the URL , so can you tell me the URL? @KonradIT

KonradIT, 2017-03-13:
@WanXiangGG I don't know, I'd put /data/data/com.termux/files/continue.sh as an URL but I am not sure. I don't have a device to test.

KonradIT, 2017-03-13:
@WanXiangGG use https://openpilot.comma.ai as the URL if you want to use Comma AI's Open Pilot.

WanXiangGG, 2017-03-13:
@KonradIT 
GET ! 
THANK YOU!

KonradIT, 2017-03-13:
@WanXiangGG you can close this issue now.

closed 2017-03-13 by WanXiangGG