commaai/neo #19: How do I get a Q4 of GIANTpanda

leo333, 2017-06-05:
How can I get a  Q4panda, GIANTpanda，Is GIANTpanda used to connect accord?

geohot, 2017-06-05:
GIANTpanda is a carrier board for the comma.ai panda, and it has the hardware needed to interface with accord, yes.

WaliedA, 2017-06-05:
Fantastic George, our issue is solved, can you advice on specific accord model we can test with as we are developing a GiantPanda board

geohot, 2017-06-05:
We worked with the 2016 Accord Touring. The braking and cruise are the same as the Acura, the steering is over K-Line.

closed 2017-06-05 by geohot

WaliedA, 2017-06-11:
George, do we need Panda to use with GiantPanda for Honda Accord on connect directly to OBDII K-line

edsammy, 2017-06-13:
Yes you need a panda. GiantPanda is simply a breakout board for the panda. It lets you mount a panda in a neo.