commaai/neo #23: Xiaomi smartphone instead OnePlus3 ?

popurax, 2017-08-03:
- Xiaomi mi5s
- Xiaomi mi6
Can I use these instead of OnePlus3 ?

Aaahh, 2017-08-03:
Yeah looking at the sources it should be easy adding any device
I'll try doing it on a fork next week
Side note commaneos should be rebased

popurax, 2017-08-06:
😎

Aaahh, 2017-08-06:
What does that mean
I haven't started yet but I was going to tomorrow

olmerg, 2017-08-09:
Can I use Onpelus5 instead of Oneplus3?

closed 2017-08-31 by popurax

rebroad, 2018-01-06:
@popurax I also need to know if Oneplus5 or Oneplus5T can be used instead. Is there an FAQ or stackexchange? Should one open a new issue to ask this question?

bertramt, 2018-01-08:
geohot has said more than once comma has no plans of supporting the 5 or 5T.  Basically if you want anything but the oneplus 3 or 3T, you are on your own.  My limited understanding is the visiond binary is very closely tied to the camera on the 3/3T and won't work on anything else.