commaai/neo #24: can anyone upload Solidworks files for given 3d STL files??

hemangjoshi37a, 2017-09-03:


Krishtof-Korda, 2017-09-03:
You can import an stl file and solidify it in solidworks with surfacing tools such as surface offset with zero offset, or thicken. If you are unsure how to do it feel free to send me the stl and I can give you back a sldprt. Cheers!

hemangjoshi37a, 2017-09-03:
Okay but that won't have solidworks supported features, that will only have
surfaces. I want to edit features..

Krishtof-Korda, 2017-09-03:
Actually it will give you exactly what you want. After you copy the
surfaces, knit everything together you can convert to solid, then you can
run feature recognition tool. You will have a solid body with features. It
won't be the original features that were modeled but you will be able to
modify the solid body and add your own features. Let me know if you need
help, just send me the stl and I can send it back as solid with features.

Cheers,

Krishtof

closed 2018-01-30 by geohot