commaai/neo #63: OBD-2 Harness schematic

JasonJShuler, 2020-09-08:
Can we please get the OBD-II harness connector schematic added?

Thanks!

ErichMoraga, 2020-09-08:
It is there already.

robbederks, 2020-09-15:
The OBD-II harness schematic (the one with GMLAN) is closed source.

As a brief general description though:
It's basically a conversion between the OBD-II spec and the OBD-C spec, consisting of:
- It connects all 3 possible CAN buses (see [this pinout](https://miro.medium.com/max/484/1*UfvsgwapNIg4fpO0iz3xBg.png)) in order to bus 0, 1 and 2 on the OBD-C side
- It detects ignition and converts that to ignition for the OBD-C spec
- It implements a mean of orientation detection for the OBD-C spec
- It has an active bi-directional conversion circuit between the GMLAN pin on the OBD-II connector and CAN bus 3 of OBD-C. This conversion circuit is powered only when the `relay power` is enabled from the OBD-C spec

Let me know if you have any further questions!

closed 2020-09-15 by robbederks

JasonJShuler, 2020-09-16:
> The OBD-II harness schematic (the one with GMLAN) is closed source.
>
> As a brief general description though:
> It's basically a conversion between the OBD-II spec and the OBD-C spec, consisting of:
>
> * [...]

I am attempting to develop the harness wiring for GM vehicles that have the cyber security gateway which means the OBD2 port is useless and that do not have an ascm. this bi-directional low-speed GM lan have circuit is something that would be very useful have. I have put countless hours into this project and closed sourcing something like this feels like a real slap in the face. It's not right. Is not your goal to have more cars supported? why would you hide the implementation details of a connector?

I understand that you have a business to run. But you're relying on a community to Port vehicles and then you tie our hands. I am providing you guys with free labor and you're going to close source the OBD2 adapter that contains the exact circuit that I need wow that is pretty trashy.

JasonJShuler, 2020-09-16:
My apologies for my we'll call it cranky comment which I have deleted; at the time I did not understand that the OBD-II harness was a complete replacement of the harness used for other vehicles I thought it was an attachment for that harness that's why I was ticked off sorry about that.