commaai/neo #67: Bosch Harness Camera connector doesn't exist?

arnarhelgi, 2021-08-19:
JAE MX34M125SF1 does not exist as far as I can see, and can't be found on mouser / digikey or even in the JAE catalog, the part number seems to be invalid and the 12 pin connector from this series does not match that connector. Is there another part  that's available?

csouers, 2021-08-22:
It's not available via retail. You have to contact JAE sales directly.

arnarhelgi, 2021-08-22:
I see, I assume that JAE will only deal in bulk? I need one, maybe two make/female connectors and pins. I’ll try them, or hope and pray that my civic has a nidec camera unit… 😬

csouers, 2021-08-22:
Based on how comma was able to source them, yeah. Bulk seems to be the key.

vanillagorillaa, 2021-08-23:
Looks like there is a typo in the diagram, will update accordingly  The correct part number is MX34M12SF1. Mouser has 820 in stock and will sell single quantities.

closed 2021-08-23 by vanillagorillaa

vanillagorillaa, 2021-08-23:
Fixed in 92b9dc17b63c4b0328dfe4ce6999eeced0f1c992