commaai/neo #69: Diagram error or misunderstanding on J533

actuallylemoncurd, 2022-06-20:
Hello, on the J533 Diagram, the pinout for "To Adapter Board" is what I assume to be the car harness j533 plug.

The counting and display of those pins are wrong.
[image]

theyre going
1 3 5
2 4 6
when its supposed to be 
1   2   3
11 12 13

Please look at how the pinout for "TO j533" is displayed as that is how VW counts their pins on the car harness side

vanillagorillaa, 2022-06-21:
The adapter board has two connectors on it, 5018762040 and 966871-1. 966871-1 number scheme is 1 2 3, 11 12 13, but the numbering scheme for 5018762040 is 1 3 5, 2 4 6. The J533 harness connects to the adapter board with connector part number 5016462000 which is why it is on the harness schematic this way. 

You can find the adapter board here, https://github.com/commaai/neo/tree/master/car_harness/boards/VW

closed 2022-06-21 by vanillagorillaa

actuallylemoncurd, 2022-07-01:
what is the point of the "adapter board?"

[image]

The J533 harness from your shop does not use this "adapter board" but rather a normal wire-wire plug

I am trying to find the pinout used for the j533 DEV harness sold on your shop.
I already have one installed but I'd like to know which buss's are filtered/tapped/etc

actuallylemoncurd, 2023-05-11:
hello, 5016462000 is the PN listed in the PDF, not  5018762040.

i only noticed this after buying  5018762040........

the PCB counterpart of 5016462000 is 5018762640

actuallylemoncurd, 2023-05-12:
wait now im confused. mouser says 5018762040 is the right mating connector. i have confirmed by PN that 5018762040 is what i ordered, and it is NOT the same as the one on my comma PCB

@vanillagorillaa did you guys change the production J533 connector?
if so, can you update the PDF or drop the new PN here?

i am trying to make a module to add some functionality to OP

actuallylemoncurd, 2023-05-12:
yeah just triple checked,  5018762040 is what i bought, and it is different than that of the connector on my genuine comma ai j533 adapter PCB from the harness purchased with my C3.

any help?