commaai/neo #70: CAN number count mismatch

actuallylemoncurd, 2022-06-20:
Comma pedal pinout shows it is "CAN1" mean while on the official relay pinout it is "CAN0"

Seems there is a discrepency of which number was used to count from, 0 vs 1

Pedal:
[image]

Relay:
[image]

ErichMoraga, 2022-06-20:
As a side note, Pedal and Giraffe were used fake ethernet spec v1, whereas the Harness [relay] Box and CPv2 (Comma Power version two) use fake ethernet spec v2.

vanillagorillaa, 2022-06-21:
Not really an issue, other older hardware in this repo also have this same discrepancy. As long as the single CAN bus on the pedal is routed to pins 3/6 on ethernet connector it will work.

closed 2022-06-21 by vanillagorillaa