commaai/neo #73: Car unsupported dashcam mode

adub8923, 2023-01-28:
I'm sure I'm not the first person to have this issue, so I apologize if this post is redundant.  Just installed openpilot on my 2022 Lexus RX.  Upon load, says car unsupported dashcam mode.  I'm not a software engineer and don't have any programming experience, but I'm able to follow instructions if someone can please provide me a solution.  Thanks!

ErichMoraga, 2023-01-28:
> I'm sure I'm not the first person to have this issue, so I apologize if this post is redundant.  Just installed openpilot on my 2022 Lexus RX.  Upon load, says car unsupported dashcam mode.  I'm not [...]

See my pinned fingerprinting related message in Comma Discord's #toyota-lexus channel.

incognitojam, 2023-01-28:
note this repo is for the comma neo, not [openpilot](https://github.com/commaai/openpilot)

closed 2023-01-28 by incognitojam